penglezos/device_xiaomi_raphael #9: There is a bug, after flashed LineageOS 20, the recovery can not start

KudouShinnyan, 2023-01-20:
The OS I was using was PixelExperience Plus 13, at this time flash LineageOS recovery works fine, then after I used LineageOS recovery to flash linage-20.0-20230117-UNOFFICIAL- raphael.zip, I tried to reboot to recovery and it started fastboot directly, long press volume up and power button also fastboot, at this time whether it is reflash LineageOS recovery or TWRP can not change the status quo, then in fastboot input fastboot boot (LineageOS recovery file.img) did not smoothly start LineageOS recovery, but input fastboot boot (twrp file.img) could smoothly start TWRP, and when I flash PixelExperience Plus 13 again, I could use the regular method to Start Recvery, and after I reflash Lineage 20.0 again, it is the same as described just now

penglezos, 2023-01-20:
Hi,
Thanks for submitting your issue. Did you follow the installation steps exactly? Formatting/wiping data from recovery? Typing fastboot -w after the described issue? Also on the previous rom your system was decrypted by any chance?

KudouShinnyan, 2023-01-20:
> Hi, Thanks for submitting your issue. Did you follow the installation steps exactly? Formatting/wiping data from recovery? Typing fastboot -w after the described issue? Also on the previous rom your system was decrypted by any chance?

Every time I flash a different ROM, I format all the partitions
I've reinstalled hundreds of apps on PixelExperience Plus 13, I'll try fastboot -w when I have time

penglezos, 2023-01-20:
Please do so and don't forget to read and follow the installation instructions as described. Let me know.

user456789, 2023-01-22:
I had the same problem following the steps exactly. Now I flashed a different recovery and everything works fine.

penglezos, 2023-01-22:
That's strange. I'm always following these steps without any problem.

closed 2023-02-03 by penglezos