penglezos/device_xiaomi_raphael #10: Improve Speaker Output

parth111999, 2023-01-21:
Can you please check what's up with right speaker output. It got recently fixed in latest VoltageOS you can diagnose the issue using this video.
https://youtu.be/6TWJaFD6R2s

penglezos, 2023-01-22:
Just tested it can confirm the right channel on mono audio is broken. Not sure how this can affect day to day usage. I have to debug it myself since VoltageOS is source closed.

ajay230ps, 2023-09-24:
Can you check this. It might help

https://github.com/Pc1598/kernel_xiaomi_raphael/commits/dev/DSP[image]

ajay230ps, 2023-09-25:
Comment from one of the dev.

[image]

penglezos, 2023-09-30:
Fixed properly on the latest release.

closed 2023-09-30 by penglezos