penglezos/device_xiaomi_raphael #12: Recovery not starting up after flashing with fastboot

fcc48, 2023-02-23:
I flashed recovery (22-Feb)  with fastboot but it is not starting when rebooting and Volume-Up.
After ~15 secs pressing Volume-Up i get black screen. 
Installed on the phone is MIUI 12.0.5 flashed with fastboot.

Maybe you have an idea what could be reason?

Additional notes:
I have in general problems flashing this device - i only can flash TWRP recoveries, Orangefox i cannot get to work.
Android 13 Roms i did get working. Result is  usually bootloops and no recovery can be loaded nor the system is starting.
Android 10 (crDroid) i got working.

Maybe something general is broken on the device.

fcc48, 2023-02-23:
Update:
Latest OrangeFox is running (i only tries an older version what was recomended for Android 13 (OrangeFox-R11.1_2_FBEv1-Unofficial-raphael.zip)) but could not flash custom ROMS with it

penglezos, 2023-02-23:
Hi,
Thanks for submitting your issue. Can you try executing `fastboot -w` (this will wipe your data, make a backup) and retry to load the recovery? What you've just described seems weird.

fcc48, 2023-02-24:
Yes, it is weird ;)

I did fastboot -w, but no success flashin the recovery.

Here the output of "fastboot -w"

`
Erasing 'userdata'                                 OKAY [  2.428s]
mke2fs 1.46.2 (28-Feb-2021)
Creating filesystem with 29136891 4k blocks and 7290880 inodes
Filesystem UUID: 8f6d1fb6-b40c-11ed-b4a7-a7b4d51766b1
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872

Allocating group tables: done
Writing inode tables: done
Creating journal (131072 blocks): done
Writing superblocks and filesystem accounting information: done

Sending 'userdata' (544 KB)                        OKAY [  0.019s]
Writing 'userdata'                                 OKAY [  0.003s]
Erasing 'cache'                                    OKAY [  0.014s]
mke2fs 1.46.2 (28-Feb-2021)
Creating filesystem with 65536 4k blocks and 65536 inodes
Filesystem UUID: 8f902e20-b40c-11ed-807c-7d8051aa9f2c
Superblock backups stored on blocks:
        32768

Allocating group tables: done
Writing inode tables: done
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done

Sending 'cache' (56 KB)                            OKAY [  0.014s]
Writing 'cache'                                    OKAY [  0.005s]
Erasing 'metadata'                                 OKAY [  0.006s]
Erase successful, but not automatically formatting.
File system type raw not supported.
Finished. Total time: 2.768s
`

penglezos, 2023-02-25:
The output shows that everything is formated successfully. Not sure what's going on from your initial issue message I suspect there is a problem with system encryption. I assume you followed the steps I provide. Are you flashing any decrypt zip (DFE) ?
Are you flashing a custom kernel or any other mod? Is your previous ROM decrypts by default or sets data to F2FS? Additional feedback is appreciated.

Thanks.

fcc48, 2023-02-26:
The device was rooted and flashed custom ROMS already 2y ago i think.
But it was not used for a year or so and i started in the last couple of weeks ;)
Also with decryption... 
I tried once with data as F2F", but reversed to EXT4.

As starting point to flash your ROM i flashed the fastboot version of MIUI (i tried Android 9 and 11 based MIUI).
Flashing these ROMS always works, and the ROM is working well. My assumtion is after flashing these ROMS everything is back to default.

Flashing custom recoveries is only successful for OrangeFox-raphael-stable@R11.1_4 and TWRP.

penglezos, 2023-02-26:
Kindly try with encryption and data formated on EXT4 file system.

fcc48, 2023-03-02:
I tried that several times after flashing MIUI stock ROMS (different Version) but recovery never was working.
You can close the ticket - in the meantime i bricked the device. I used MiFlash and flashed a stock ROM. But now i cannot unlock it anymore because SIM cards are no longer detected...  ;)
Thanks for your great work for the device and your help!

closed 2023-03-02 by penglezos