penglezos/device_xiaomi_raphael #14: UDFPS not working for banking app

yeray142, 2023-03-12:
I installed your ROM yesterday and I noticed that I can't set my fingerprint as the unlocking method on my Korean banking app 우리WON뱅킹 (Woori Bank). 

When setting the fingerprint as the unlocking method it says that it cannot verify it correctly. I'm uploding the corresponding logs. It was working properly on latest MIUI. 

Tell me if you need any further information.

NOTE: The rest of apps/banking apps are working properly with the fingerprint sensor. I'm NOT using any other Kernel/Recovery/MOD, etc...

[log.txt](https://github.com/penglezos/device_xiaomi_raphael/files/10951333/log.txt)

penglezos, 2023-03-13:
Thanks for your submission. On my bank application I can register and use the fingerprint authentication method successfully, uppon log checking maybe there's an application problem?  Can you share a video of how the application responds to the fingerprint authentication?

yeray142, 2023-03-14:
Sure! As the app doesn't allow third party recording apps to record the app itself, I used my laptop webcam, I apologize for the video quality.

Also, the app uses AhnLab V3 Mobile Plus (it's some type of antivirus) everytime the app runs for security reasons. Don't know if that can affect somehow the fingerprint sensor.

Video: [Link to YouTube](https://www.youtube.com/watch?v=Ew-v-R-mmTU)

The error message that appears when I cancelled the fingerprint registration says something like: "The registration was cancelled by the user" (in Korean). Sometimes when registering the fingerprint in that same app, it says something like "Press firmly".

penglezos, 2023-04-02:
April update might fix your specific app issue.

penglezos, 2023-04-17:
@yeray142 please flash latest release and let me know. Most likely it's an app issue but I spent some time work-arrounding this.

yeray142, 2023-04-17:
I flashed latest update through OTA and it's still not working. Maybe it's an app issue.

penglezos, 2023-04-17:
It seems so, report to application developers.