penglezos/device_xiaomi_raphael #18: No dialing tone when starting a call

deify, 2023-04-25:
So I don't get any dialing tone in a call. It starts with one single sound and then there is total silence until the called Person picks up or dismisses the call.

Same for VOLTE and WifiCall. Havent tried using non VOLTE calls. On the stock ROM everything worked fine.

If i can help debugging this tell me and i will try to assist!

Best regards

closed 2023-05-06 by penglezos

deify, 2023-05-12:
Hey, i've installed the latest release today. This issue dies not seem to be fixed within this release. Did you find a fix that will be included in a later release, or do you need and hell debugging the issue?

Besteht regard

penglezos, 2023-05-15:
Hi, maybe its carrier dependant? I've seen couple reports online on devices like Google Pixel, Oneplus and Xiaomi. I've tested that personally on iPhone 12, Redmi Note 6 pro and Google Pixel 6a none of these devices had ringback tone when VoLTE functionality was enabled on my carrier. Any suggestions is welcomed.