penglezos/device_xiaomi_raphael #23: System updater behavior

da3dsoul1, 2023-07-27:
I'm on a latest build, always update via adb sideload, just want to make sure if what I'm experiencing is normal or not. In system updater when I try to disable Update recovery it shows a toast that's it's impossible to do so, is it a bug or intended?
[image]

penglezos, 2023-07-27:
You should be able to disable that. As it seems the Updater application lacks of internet permissions make sure you enable em.

da3dsoul1, 2023-07-27:
Enabled permissions, update checking is working, but still can't disabled update recovery function for some reason

penglezos, 2023-07-27:
What recovery do you have installed LineageOS or something else?

da3dsoul1, 2023-07-27:
> What recovery do you have installed LineageOS or something else?

I use ofox that supports fbev1

penglezos, 2023-07-27:
This feature only works with LineageOS recovery correctly.

closed 2023-07-27 by penglezos

da3dsoul1, 2023-07-27:
> This feature only works with LineageOS recovery correctly.

Flashed lineage recovery,  same thing

penglezos, 2023-07-27:
[image]

You are obviously doing something wrong.

da3dsoul1, 2023-07-27:
Weird, gonna apply update from lineage recovery, if it doesn't work then I guess I have to clean flash

da3dsoul1, 2023-07-27:
Okay, reapplying update through lineage recovery fixed the issue, gonna keep using it then. How stable it is when updating via Updater app? Both system and recovery updates I mean