penglezos/device_xiaomi_raphael #25: ViLTE (Video over LTE)

penglezos, 2023-08-20:
ViLTE (Video over LTE) is not working. Option does not appear at dialer at all when enabled from settings.

ajay230ps, 2023-10-02:
Hi Englezos,

ViLTE  works on CrDroid ROM (29 Sep build) for raphael. It words on both stock dialer and google dialer.

I'm using Airtel sim (India). To make it work first I need to make a normal call to Airtel sim user, once he picks then click on Video to converts to this to video call, once the other party picks up the video call then it works.

Hope this helps.

penglezos, 2023-10-02:
Hi @ajay230ps,

Thanks for your input. Can you also try and report if ViLTE works on my builds?

ajay230ps, 2023-10-06:
Sure. I'll check in next few days and report back here.

ajay230ps, 2023-10-07:
ViLTE is not working on both stock and google dialer.