penglezos/device_xiaomi_raphael #30: ‘Prevent accidental wake’ feature not working

monkydluffy25, 2024-07-26:
Prevent accidental wake up not working on my device.
Can you confirm it is same for you or just me?

rPAUL0, 2024-07-26:
> Prevent accidental wake up not working on my device. Can you confirm it is same for you or just me?

Work perfectly for me.

penglezos, 2024-07-26:
> Prevent accidental wake up not working on my device. Can you confirm it is same for you or just me?

How exactly did you test it? It works for me.

monkydluffy25, 2024-07-26:
I just covered the top part of screen before turning on the screen to test it. Proximity sensor is working fine as screen turns of during calls.

I did not modify anything (flashed rom + bitgapps).

If it is working for everyone then mine might have some random issue.

closed 2024-07-26 by penglezos