penglezos/device_xiaomi_raphael #31: Pnb One banking app login problem

rPAUL0, 2024-07-26:
For me a banking app called "PNB One" this app not working properly, when I to login the app it's show "Unable to process your request. Please try later"- this popup show. I found this banking app work only specific rom which don't have "init.rc issue"(it's mean in momo app Los21 show 'init.rc is modified'), most probably "init.rc" causing this banking app login issue.

l also generate log, I don't know it's captured in log or not.
[image]
[logcat.txt](https://github.com/user-attachments/files/16389138/logcat.txt)
[radio.txt](https://github.com/user-attachments/files/16389143/radio.txt)

penglezos, 2024-07-26:
"app not working properly, when I try to login the app it's show "Unable to process your request." According to your own works it seems like the problem is the application itself. Check their Google Playstore reviews many people have the exactly issue like yours.

Ref: https://play.google.com/store/apps/details?id=com.Version1&hl=en_IN

Also are you rooted? This application is outdated and unreliable for those kinds of checks.

rPAUL0, 2024-07-26:
> "app not working properly, when I try to login the app it's show "Unable to process your request." According to your own words it seems like the problem is the application itself. Check their Google Playstore reviews many people have the exactly issue like yours.
> 
> Ref: https://play.google.com/store/apps/details?id=com.Version1&hl=en_IN
> 
> Also are you rooted? This application is outdated and unreliable for those kinds of checks.

After it's show in los I go back and install votlateos for confirm it's app issue or rom issue, and i see this app working fine in voltageos

penglezos, 2024-07-26:
> After it's show in los I go back and install votlateos for confirm it's app issue or rom issue, and i see this app working fine in voltageos

Are you using magisk to root?

rPAUL0, 2024-07-26:
> > > "app not working properly, when I try to login the app it's show "Unable to process your request." According to your own words it seems like the problem is the application itself. Check their Google Playstore reviews many people have the exactly issue like yours.
> > > Ref: https://play.google.com/store/apps/details?id=com.Version1&hl=en_IN
> > > Also are you rooted? This application is outdated and unreliable for those kinds of checks.
> > 
> > 
> > After it's show in los I go back and install votlateos for confirm it's app issue or rom issue, and i see this app working fine in voltageos
> 
> Are you using magisk to root your system?

No sir, I use with stock and mind the gapps installed

penglezos, 2024-07-26:
> No sir, I use with stock and mind the gapps installed

You obviously have init.rc modified either by a magisk module or you did it manually.

For the shakes of my sanity I installed that crap app without being rooted and here are the results:

[image]

rPAUL0, 2024-07-26:
> > > > > "app not working properly, when I try to login the app it's show "Unable to process your request." According to your own words it seems like the problem is the application itself. Check their Google Playstore reviews many people have the exactly issue like yours.
> > > > > Ref: https://play.google.com/store/apps/details?id=com.Version1&hl=en_IN
> > > > > Also are you rooted? This application is outdated and unreliable for those kinds of checks.
> > > > 
> > > > 
> > > > After it's show in los I go back and install votlateos for confirm it's app issue or rom issue, and i see this app working fine in voltageos
> > > 
> > > 
> > > Are you using magisk to root your system?
> > 
> > 
> > No sir, I use with stock and mind the gapps installed
> 
> You obviously have init.rc modified either by a magisk module or you did it manually.
> 
> For the shakes of my sanity I installed that crap app without being rooted and here are the results:
> 
> ![Screenshot_20240726-123552_Momo](https://private-user-images.githubusercontent.com/7030531/352486641-84f0ee84-9292-459e-9527-a9ad959719f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5ODc0NjcsIm5iZiI6MTcyMTk4NzE2NywicGF0aCI6Ii83MDMwNTMxLzM1MjQ4NjY0MS04NGYwZWU4NC05MjkyLTQ1OWUtOTUyNy1hOWFkOTU5NzE5ZjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjZUMDk0NjA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjk2ZmQ5ZTk3M2YzYmZlYjg1NzAzYWQxMzEwOTM3ZjFlZjAxMjMyMjlkNDdjODdhZDliNjVhNDU3OWUxMWM1ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.PzKQXPPSMVMnpR0TnY4_0CIJkzAAd9mSV1kn-j9peIw)

Okay, it's strange i did't install magisk. let's cleanflash and see the problem fixed or not

rPAUL0, 2024-07-26:
> You obviously have init.rc modified either by a magisk module or you did it manually.
>
> For the shakes of my sanity I installed that crap app without being rooted and here are the results: [...]

After clean flash still the same problem persist :)

penglezos, 2024-07-26:
> Unable to process your request

Expected the application is broken.

rPAUL0, 2024-07-26:
> > Unable to process your request
> 
> As expected the application is broken.

Ok sir

closed 2024-07-26 by penglezos