penglezos/device_xiaomi_raphael #32: Vi Sim call problem

rPAUL0, 2024-07-26:
When I call someone with my "Vi Sim" there is no ringing sound appear, but call connected with other side, in my "Jio sim" work properly.
It is a specific sim and rom problem, in some rom (Like crdroid, Derfest, _voltage os also fix this in a14_) ringing sound appear properly.

[radio.txt](https://github.com/user-attachments/files/16389679/radio.txt)
[logcat.txt](https://github.com/user-attachments/files/16389682/logcat.txt)

penglezos, 2024-07-26:
There is no ring-back tone on 4G when VoLTE is enabled on both sides. Sometimes its also carrier dependant.

rPAUL0, 2024-07-26:
> There is no ring-back tone on 4G when VoLTE is enabled on both sides. Sometimes its also carrier dependant.

I notice something when I turn off volte then ring sound work fine, I call in customer care and then said they don't know.
I also see in reddit and oneplus comunity some oneplus user says they also suffer with this in stock oneplus 6/7, but after the update it's fix

penglezos, 2024-07-26:
> I notice something when I turn off volte then ring sound work fine, I call in customer care and then said they don't know. I also see in reddit and oneplus comunity some oneplus user says they also [...]

I already gave you my answer that's how it works.

rPAUL0, 2024-07-26:
> I already gave you my answer that's how it works.

Ok sir, thank you

rPAUL0, 2024-07-26:
> Ok sir, thank you

Does this problem fix with some network upstream ?