penglezos/device_xiaomi_raphael #34: LSPosed

heloonihao, 2024-07-29:
My lsposed1.9.2 interface not work,But he can still detect the module and Send notifications
[image]

penglezos, 2024-07-29:
From the Bug report section which clearly you did not read:

> Support is NOT provided if you: installed a custom recovery, decrypt your /data, installed a custom kernel, rooted your /system and installed any other mods.

heloonihao, 2024-07-29:
[crash-zygote64-29_07-16-44-32_385.zip](https://github.com/user-attachments/files/16410006/crash-zygote64-29_07-16-44-32_385.zip)
here is log，using kernelsu（magisk is same crash），kernelsu only install zygisknext， zygisk-lsposed 1.9.2，android kernel using SOVIET-STAR

heloonihao, 2024-07-29:
sorry about that

closed 2024-07-29 by heloonihao