penglezos/device_xiaomi_raphael #36: Random Reboots on lineage-21.0-20240723 build

MiZantr0pe, 2024-08-23:
Hi, [penglezos](https://github.com/penglezos)! 

I've been using the firmware for about a month. I installed it according to your instructions, used Lineage recovery, mindgaps, without installing additional magisk modules and other root stuff. During this time, there have already been 3 random reboots - one while using one of the messengers when starting the keyboard for typing, two reboots occurred in a state of rest, when the phone was just lying next to me on the table.

I attached the logcat.txt file, but I made it right after the phone rebooted. Sorry, but I don't know how to make a file that will show the problems that preceded the reboot of the device. Can you tell me how I can do this next time?

The guys from the 4pda forum also have reboots. Some have 5-7 reboots a day, while others (like me) - only after 100+ hours of stable operation.

[logcat.txt](https://github.com/user-attachments/files/16726381/logcat.txt)

penglezos, 2024-08-23:
Hi, thanks for your report!

First of all, you are not on the latest release, so please make sure you update accordingly. Regarding the "random reboots" from the logcat you posted I can't find anything problematic that can cause sudden reboots. I've been driving my LineageOS 21 builds daily for months without "random reboots" but we need to monitor it for sure. Did you by any chance changed the default kernel? are you encrypted or decrypted? did you flashed "legacy to dynamic" zip? Can you also provide the installed apps you have? Maybe you have an application that crashes the system I've seen that in the past with Skype (strange I know).

> The guys from the 4pda forum also have reboots. Some have 5-7 reboots a day, while others (like me) - only after 100+ hours of stable operation.

I don't know what 4pda is neither I can accept that as a form of report.

MiZantr0pe, 2024-08-23:
> First of all, you are not on the latest release

Ok, about an hour ago I updated to the latest Lineage rom version. I'll watch its behavior.

> Did you by any chance changed the default kernel?

Nope, there is default kernel, default lineage recovery from your installation guide and mindthegapps-package

> are you encrypted or decrypted?

Im see "Encrypted" in "Encryption & Credentials" menu, and also I didn`t flash "legacy to dynamic" zip - because it wasn't in your installation guide, is it correct or not?

> Can you also provide the installed apps you have? 
Ok, let`s see files attached

[screenshots.zip](https://github.com/user-attachments/files/16728591/screenshots.zip)

penglezos, 2024-08-23:
Yes you are encrypted. Holy shit there are so many apps installed. Let see how this release goes for you.

MiZantr0pe, 2024-08-23:
> Holy shit there are so many apps installed.

These screenshots also have system apps :D and I forgot to delete two pictures that duplicate the others

Can you tell me how I can export a list of applications as a TXT file without having to create a lot of screenshots?

> Let see how this release goes for you.

You mean I should look at the behavior of the latest ROM version from August 20th? OK

> but I don't know how to make a file that will show the problems that preceded the reboot of the device.

Can you please tell me how I can create a logcat.txt file next time so that it shows the problems that happened BEFORE rebooting the smartphone?

Tfviktor, 2024-08-23:
I also have random reboots on 20240723.

MiZantr0pe, 2024-08-23:
@Tfviktor hello! Let's help the developer - please answer his previous questions to me

Lazarus00001, 2024-08-24:
I also faced random reboot previous build and latest build also, I didn't report because i didn't find any log on /sys/fs/pstore/

penglezos, 2024-08-31:
I think it's safe to say that after 8 plus days of consistent uptime (on this test) that there are NO random reboots occurred.

[image]

MiZantr0pe, 2024-09-01:
> I think it's safe to say that after 8 plus days of consistent uptime (on this test) that there are NO random reboots occurred.

Hello! I have the same, 200 hours of stable work. But I'm not sure about stability yet - on the previous version the first reboot occurred after 500 hours uptime.

penglezos, 2024-09-01:
> > I think it's safe to say that after 8 plus days of consistent uptime (on this test) that there are NO random reboots occurred.
> 
> Hello! I have the same, 200 hours of stable work. But I'm not sure about stability yet - on the previous version the first reboot occurred after 500 hours uptime.

Understandable my recent screenshot was just an example. I've been running my LineageOS 21 builds since mid November with more hours uptime without any random reboots whatsoever. 20 days of uptime on your previous result is in fact an indication of stability in my opinion. Based on the other non valid reports I can't do much.

MiZantr0pe, 2024-09-02:
@penglezos After ~210 hours of work, the smartphone rebooted last night. It was just lying next to me, I was not using it at that moment. It was a complete reboot with the display of the Lineage OS logo.
Please tell me how I can provide you the necessary information for analysis? The events in the logcat.txt file I created now no longer include events before the reboot that happened last night.

penglezos, 2024-09-02:
> @penglezos After ~210 hours of work, the smartphone rebooted last night. It was just lying next to me, I was not using it at that moment. It was a complete reboot with the display of the Lineage OS [...]

That's sad to hear. Can you grab console-ramoops logs from /sys/fs/pstore ?

MiZantr0pe, 2024-09-02:
> That's sad to hear. Can you grab console-ramoops logs from /sys/fs/pstore ?
> 
> `adb shell cat /sys/fs/pstore/console-ramoops-0 > console-ramoops.txt`

@penglezos I did what you said using adb:

cat: /sys/fs/pstore/console-ramoops-0: No such file or directory
^ the result of the command in the console looks like this. Am I doing something wrong?

penglezos, 2024-09-02:
> @penglezos I did what you said using adb:
>
> cat: /sys/fs/pstore/console-ramoops-0: No such file or directory ^ the result of the command in the console looks like this. Am I doing something wrong?

@MiZantr0pe and everyone willing to debug this please flash this kernel image in fastboot:

Link: [https://we.tl/t-kTuGZNb91N](https://we.tl/t-kTuGZNb91N)

- `fastboot flash boot boot.img`
- restart phone
- verify that kernel is installed
- reboot to system once again
- verify that console-ramoops log is captured: `adb shell cat /sys/fs/pstore/console-ramoops-0 > console-ramoops.txt`

Next time you face "random reboot" immediately capture and send the log here via this command:
`adb shell cat /sys/fs/pstore/console-ramoops-0 > console-ramoops.txt` or `adb shell cat /sys/fs/pstore/console-ramoops > console-ramoops.txt`

@MiZantr0pe in your case unfortunately this log never generated. Please if it's possible, flash this kernel run the command and send me the results, we might get lucky and get any suspicious activity resulting into your random reboots.

Thanks for your time and report I'm trying to figure out why this happens to only limited amount of people but I don't have enough feedback.

MiZantr0pe, 2024-09-02:
@penglezos can you explain this boot.img that needs to be installed, - what`s the difference between the standard boot.img? Is it suitable for everyday use and can I not be afraid that it will not cause unstable work by itself? Because earlier you don`t recommended change ROM`s kernel 😅

penglezos, 2024-09-02:
> @penglezos can you explain this boot.img that needs to be installed, - what is the difference between the default boot.img? Is it suitable for everyday use and can I not be afraid that it will not cause unstable work by itself? Because earlier you dont recommended change ROM's kernel 😅
> 
> > * verify that kernel is installed
> 
> How can I verify this?

About phone > Android version

MiZantr0pe, 2024-09-03:
@penglezos Unfortunately, capturing the log file does not work on your kernel, screenshots are attached

[image]
[image]

Lazarus00001, 2024-09-03:
@MiZantr0pe after flash boot.img and reboot into system enable usb debuging and after that u need to wait until random reboot happen again, then try to capture ramoops log

MiZantr0pe, 2024-09-03:
@penglezos 

> u need to wait until random reboot happen again

I have another one reboot - no logs again!

[image]

penglezos, 2024-09-03:
> @penglezos Unfortunately, capturing the log file does not work on your kernel, screenshots are attached
> 
> ![a2710c9c-b010-4513-84cc-1ce9e6e640db](https://private-user-images.githubusercontent.com/66994182/363890760-4eb0a80c-a7ac-4e67-be6a-b0c2b573057b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjI2MjMsIm5iZiI6MTcyNTM2MjMyMywicGF0aCI6Ii82Njk5NDE4Mi8zNjM4OTA3NjAtNGViMGE4MGMtYTdhYy00ZTY3LWJlNmEtYjBjMmI1NzMwNTdiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAzVDExMTg0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcyNDYzM2FmYjk1M2Y5NTA5MjM5NjNmMmIwNTNkNTY0MjgxMmI5Yjc0NzI5YzMxYmM1NzA3NmYxOGJjZTgyNTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6HbzcLl61bdLO1QDXl4WKDTwcfcZh5ijngeqIxniDl8)
> 
> <img alt="2024-09-03_10-52" width="841" src="https://private-user-images.githubusercontent.com/66994182/363890783-607ec71b-f7f9-471a-9005-59cda21930f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjI2MjMsIm5iZiI6MTcyNTM2MjMyMywicGF0aCI6Ii82Njk5NDE4Mi8zNjM4OTA3ODMtNjA3ZWM3MWItZjdmOS00NzFhLTkwMDUtNTljZGEyMTkzMGYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAzVDExMTg0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3YjA2YTJhMWQxZjUxMGEwZmM4ZGNkMzg3YTcxNjhmNzQxMWE2NTgxMGIwMzlhZTA1ODU4MTEwNTgyMmRiZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JxphQRrKrYM5V4V0v8XBkQO9E8a6diYwsRsKl0tCWRo">

@MiZantr0pe 

I assume you have adb debugging enabled and latest adb & fastboot tools installed.

Try to enable "Rooted debugging" from the developer settings.

Open command line and type:

- adb root
- adb remount
- adb shell ls /sys/fs/pstore

what is the output?

MiZantr0pe, 2024-09-03:
> what is the output?

just nothing :(

[image]

penglezos, 2024-09-03:
> * adb shell cat /sys/fs/pstore/console-ramoops-0 > console-ramoops.txt

It should look like this:
[image]

and the log file:

[image]

I don't know what's wrong with your setup mines as you can see works. Are you using  the latest SDK Platform-Tools from https://developer.android.com/tools/releases/platform-tools or something else?

penglezos, 2024-09-03:
> > . Are you using the latest SDK Platform-Tools from https://developer.android.com/tools/releases/platform-tools or something else?
> 
> @penglezos I just downloaded platform-tools for windows from your link. Disabled developer options on the phone. Rebooted the phone. Enabled developer options again. Enabled the root debugging flag. Launched the terminal in the platform-tools folder, entered your 3 commands - the result did not change. Screenshots in the attachment.
> 
> ![image](https://private-user-images.githubusercontent.com/66994182/363982018-cda369da-57fc-4768-bfa6-793153ea298a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNzQ0NDUsIm5iZiI6MTcyNTM3NDE0NSwicGF0aCI6Ii82Njk5NDE4Mi8zNjM5ODIwMTgtY2RhMzY5ZGEtNTdmYy00NzY4LWJmYTYtNzkzMTUzZWEyOThhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAzVDE0MzU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3N2ExMTkzYzY2YjgzNTdmMDYzMjMyNTcyMTljYThkNWJkZmViM2JiZDE3MGY5NzlhODhmN2I0NzI5NTUzYzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.F074qF7LVU4ACFnvu3er9oknVEayUWFVkTC8BmJw_4c)

I don't know what else to say mate mines working perfectly as you can see from the screenshots attached previously. Something is screwed at your end. Did you tried to flash stock ROM and clean flash LineageOS.

MiZantr0pe, 2024-09-03:
> Did you tried to flash stock ROM and clean flash LineageOS?

No, I didn't do that because I think your flashing guide eliminates any issues that might have been caused by the previous ROM's junk. If I install the stock miui 12 ROM using miflash tool, need I to unlock the bootloader again or not?

penglezos, 2024-09-03:
> > Did you tried to flash stock ROM and clean flash LineageOS?
> 
> @penglezos no, I didn't do that because I think your flashing guide eliminates any issues that might have been caused by the previous ROM's junk. If I install the stock miui 12 ROM using miflash tool, need I to unlock the bootloader again or not?

There's no need to reunlock the bootloader just make sure to apply the correct options. Check out the internet for instructions.

MiZantr0pe, 2024-09-04:
@penglezos ok I'll try reflash the stock ROM and install your 20th august's Lineage today

MiZantr0pe, 2024-09-09:
@penglezos ok, I re-flashed ROM v20.08.2024 over the official stock MIUI12, there were no reboots so far (~130 hours of work). I will continue to observe. I did not install your modified kernel just in case - everything is by default. I won't install the latest version from today for now - I hope that reboots on the version from August 20 will disappear, and then I will be able to use more recent versions of the firmware with complete confidence.

penglezos, 2024-09-09:
> @penglezos ok, I re-flashed ROM v20.08.2024 over the official stock MIUI12, there were no reboots so far (~130 hours of work). I will continue to observe. I did not install your modified kernel just [...]

I advise you to update straight to the latest release.

Akshayballoria, 2024-09-11:
I m facing touch issue while on gaming when phone get hot ..touch become sluggish

barunovas, 2024-09-17:
I'm also have reboots first hour of usage. And after that - the most stable work possible.

penglezos, 2024-09-17:
> I'm also have reboots first hour of usage. And after that - the most stable work possible.

I'm sorry dude but I can't help you like that. Read the discussions on the following page and provide the necessary logs.

KrustyKrabCoder, 2024-09-21:
I encountered the same issue. In my case, the phone appeared to reboot (showing the boot logo and all), but when checking the uptime, it hadn't actually reset to zero.

I resolved this by flashing the complete Xiaomi ROM using fastboot (rather than just the firmware). You can download it here: [Xiaomi ROM for Raphael (V12.5.2.0)](https://xmfirmwareupdater.com/miui/raphael/stable/V12.5.2.0.RFKMIXM/). Simply flashing the firmware via recovery didn't seem to update all the necessary partitions, especially since I previously had LineageOS 17.1 installed, which might have corrupted some parts of the phone.

After flashing the full Xiaomi ROM and waiting for the initial boot, I rebooted into fastboot mode, installed the latest Lineage recovery, and flashed the latest LineageOS release. The phone has now been stable for over 100 hours without any issues.

I suggest modifying the instructions from "Installing device firmware from recovery" to "Installing the latest Xiaomi ROM and firmware via fastboot." This ensures a more consistent starting point for flashing LineageOS, as it updates all the phone's partitions.

Important: Make sure to use the flash_all.sh or flash_all.bat script, not flash_all_lock.sh, as that would lock the bootloader and cause delays while waiting to unlock it again.

martinpl, 2024-09-25:
Could anyone reupload debug boot.img, i still have that "reboots" but uptime don't reset for me. I have those those from LOS 20 but it don't bother me so much because the mostly happend when I don't use phone. I triedy preflashing MIUI and keeping ROM clean (no Magisk).

penglezos, 2024-09-25:
> Could anyone reupload debug boot.img, i still have that "reboots" but uptime don't reset for me. I have those from LOS 20 but it don't bother me so much because the mostly happened when I don't use phone. I tried preflashing MIUI and keeping ROM clean (no Magisk).

If you are on the latest September build there is no need to search for this kernel version, commits already present.

martinpl, 2024-09-26:
Logs: phone was ringing (phone call) and after 0.5s i saw LineageOS boot animation, uptime wasn't reset
[console-ramoops.txt](https://github.com/user-attachments/files/17149928/console-ramoops.txt)

penglezos, 2024-09-30:
> Logs: phone was ringing (phone call) and after 0.5s i saw LineageOS boot animation, uptime wasn't reset [console-ramoops.txt](https://github.com/user-attachments/files/17149928/console-ramoops.txt)

Hi @martinpl,

Finally a user provids actual logs. To clear things up if uptime does not reset that's not an actual reboot that's a SystemUI restart. Looking at the logs you provided I identified this denial spamming a lot:

`SELinux: avc:  denied  { find } for pid=4767 uid=90069 name=content_capture scontext=u:r:isolated_app:s0:c512,c768 tcontext=u:object_r:content_capture_service:s0 tclass=service_manager permissive=0`

I'm not sure how this is triggered or if it's responsible for any reboots, but I silenced it for the next release.

For the record my current uptime before I rebooted my device yesterday intentionally for flashing purposes was 500+ hours (20 days+)
and before that, while I was in August release was approximately 300 hours so more or less 30 days of stable operation.

Thanks for your time reporting this.

closed 2024-10-17 by penglezos