penglezos/device_xiaomi_raphael #37: App security lock function

giallino64, 2024-09-08:
Hi penglezos,
I recently upgraded LOS from version 20 to 21 (build 20240820) following your instructions. I am also doing very well with version 21 and everything seems to be working fine. However, I would like to report that when an application is locked with the security options, its launch is conditioned by using the fingerprint, in my personal case, while when it is opened from a notification, the app opens normally without requiring unlocking. The "problem" was present in LOS 20 and now also in LOS 21.

penglezos, 2024-09-09:
Hello @giallino64,

That's how "Protected Apps" are implemented in LineageOS. The code is implemented in Launcher instead system-wide. I know it's not the most efficient way but it is what it is until someone finds a better way.

giallino64, 2024-09-09:
Hi @penglezos
Thanks for the information and also for the work you do for the community.

closed 2024-09-09 by giallino64